MCUdude/MegaCore #226: digitalWrite does not work in flash memory when placed above 64k

felias-fogg, 2025-08-27:
I noticed that when digitalWrite (and other functions) are placed high up in flash memory, they do not work. The reason is probably that the macros used in digitalWrite make use of pgm_read_byte, which only works with 16-bit addresses. An example to illustrate that is the following little sketch:

```
const byte tab1[32767] PROGMEM = { };
const byte tab2[32767] PROGMEM = { };

void setup()
{
  unsigned long m1 = pgm_get_far_address(tab1);
  unsigned long m2 = pgm_get_far_address(tab2);
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
}
```

This sketch does not blink, but when the line `unsigned long m1` is commented out, it will blink.
Of course, this is not only a problem of MegaCore. I already reported an issue at the ArduinoCore-avr repo. Any idea how to mitigate the problem?

felias-fogg, 2025-08-27:
Actually, I found a PR to 'fix' this issue (from 8 years ago): https://github.com/arduino/Arduino/pull/6317/commits/db87e0c875379d923f71be7d8a428738cd98e999

MCUdude, 2025-08-28:
Great catch! As long as it doesn't affect the performance or compiled size when the user program is less than 64k, I'm fine with this fix. Please note that changes to the Arduino core files have to be submitted to [this repo](https://github.com/MCUdude/MCUdude_corefiles), which then again is linked to MegaCore, MightyCore, MiniCore, and MajorCore. So it's important that this doesn't affect devices with flash <=64k.

felias-fogg, 2025-09-02:
Added a PR that adds a script to check for PROGMEM overflow, i.e., avoiding a breaking change: https://github.com/MCUdude/MegaCore/pull/227