MCUdude/MegaCore #228: Potential urboot and debugger conflict

felias-fogg, 2025-09-03:
I just reread the MegaCore README and noticed that the bootloader (urboot) now also offers functionality to the user sketch (namely, writing to flash). When using a debugger, flash will be erased initially to allow for uploading the binary. The bootloader is gone, and calls to functions in the bootloader will likely lead to a restart. I could upload the code page by page, erasing each single page before programming it, but this is probably slower. Do you see any possibility of including the bootloader in the ELF file of the user sketch? 

Perhaps this could be done while linking when the "Optimize for Debugging" flag is set? However, the bootloader code would have to be removed when the HEX file is generated because otherwise, the bootloader might overwrite itself. All this does not need to be addressed right away, but this note should remind us that this might cause problems in the future.

felias-fogg, 2025-09-05:
I have implemented a page-by-page erase functionality. It is, unfortunately, significantly slower than when erasing the flash memory initially. So, this is definitely not an option.

MCUdude, 2025-09-10:
> Do you see any possibility of including the bootloader in the ELF file of the user sketch?

I'm not sure how to do this. And you can't flash a user sketch along with a bootloader binary just like that. You also need to patch the reset vector to point to the start of the bootloader and not where the application begins.

I think it's OK that debugging along with Urboot functionality is a known limitation. However, being able to "recover" after the debugging session (re-load the bootloader with a working reset vector) is important.